bigtreetech/HermitCrab #5: No Max Power Spec in Documentation

zedn4c3r, 2021-11-17:
None of the documentation lists a max power load for items attached to the tool plate.

zedn4c3r, 2022-02-08:
Part of the intended use case for this product is to be able to connect laser tool heads and other hardware to the tool plate. The selection of a laser is highly dependent on how much power it can provide.

mikerocklewitz, 2022-07-02:
I second this.

I received a hermit crab can for review purposes and i have no idea how much i can push thru it or what my max heater size can be.

The power connector for micro usb mode looks undersized, but the tool board for usb c has 5 amp fuses.....i dont know what this means

What is the max heater cartridge a person can use?

cd85233, 2022-08-12:
Posted elsewhere but I'd also like to know the 5v ratings. Most importantly the maximum current it can source.

HWiese1980, 2022-12-13:
Should definitely be! I almost grilled my Crab due to too high current

eliasmagn, 2023-01-26:
Its a joke they didnt provided any information on that till this day!
I have three hermit crab and i am considering to never buy btt stuff again !