bigtreetech/HermitCrab #9: Bypass jumpers

zedn4c3r, 2021-12-12:
Add a note in the manual to remove bypass jumpers for tool ports in use.

NAPCAL, 2022-06-15:
I have mine working with all jumpers on for the tool ports.
[image]

p-v-n, 2022-08-29:
These jumpers simply don't even make any sense! The way the whole thing is wired is as follows: 

Starting from CAN-IN plug, CANH /CANL lines go directly to A6/A7 pins (D+/D- ) respectively of female USB-C connector T0. Pins B6/B7 of this connector go to pins A6/A7 of the next USB-C connector (T1). First set of jumpers go between A6<-->B6 and A7<-->B7 pins of T0 connector. This same thing repeats for all four connectors, then B6/B7 pins of connector T3 go to CAN-OUT, while optionally connecting a terminating resistor by the the other jumper next to it.

This all would make perfect sense, *IF* USB-C cable had wires for both A and B sides of D+/D- signals. In that case, by removing the jumpers, you could make the CAN signal go to MCU through A6/A7 lines, and then return through B6/B7 lines (A6/B6 and A7/B7 are permanently connected in the receptacle on MCU side), or you could bypass the connector by leaving the jumpers on.

However, there is only one pair of wires that carries D+/D- signals in the cable, which is fine for normal USB-C functionality because the correct side will be detected by the host by looking at CC1/CC2 pins.
However, for HermitCrab case, if you remove the jumpers at connector T0, the signal will not go any further to next connector T1, and thus to terminating resistor. Also, depending on the orientation of cable plugs, may make it to MCU side, but will not be able to return (because of missing D+/D- lines for the reverse side of USB-C), or not even be able to leave the tool board.

Perhaps, I missed something, and someone can point me to where I might be wrong, but as of right now, the entire purpose of these jumpers is not really clear to me, because to make things (barely) work you must always keep them in place.

mikerocklewitz, 2022-09-10:
Schematics were uploaded the other day.

That said, it only works for me with the jumpers in place.

NAPCAL, 2022-09-10:
USB-C connector pins issue using **A6(D+)[CAN_,H], A7(D-)[CAN_L]** CAN-out from the multi-tool board and CAN-in on the Hot Mode board, and **B6(D+)[CAN_H], B7(D+)[CAN_L]** as CAN-out on the Hot Mode board and CAN-in on the multi-tool board.

While most USB-C plugs have B6(D+), & B7(D-), all cables don't have these connections (see Microchip PDF), so there will be no return to the multi-tool board. Possibly an adaptor could be made (see below) to shift the connections, but this adaptor would be needed at both ends of the cable. This is why the jumpers must be installed for the bus to be correctly terminated or continue to the CAN out connector.

Also, the Hot Mode schematics show a 120-ohm termination resistor (R17) across the CAN_H & CAN_L on that board, but that resistor is not in place on my board.

http://ww1.microchip.com/downloads/en/appnotes/00001953a.pdf

[mikerocklewitz](https://github.com/mikerocklewitz), you could make the adaptor, but you would need a second set of the breakouts you ordered.  

[image]

mikerocklewitz, 2022-09-10:
Not sure why i would need a second set of adapters if i use a usb-c cable with the right quantity of wires....plug it into the tool board and run it to the female usb-c breakout on the skr2 side of things.  Not planning on powering anything this way just gonna run can, ground, and complete the return.

NAPCAL, 2022-09-10:
It is the only way power gets to the Tool head (hot mode board)

mikerocklewitz, 2022-09-10:
> It is the only way power gets to the Tool head (hot mode board)

Im on something totally different than what you're speaking of.

Im gonna be running a skr2 via canbus plugged into the tool board. A person can add a can transceiver to the skr board......  Power will be seperate. And the hotmode board is a whole different topic.