bigtreetech/HermitCrab #11: Missing Wiring  harness Instructions for Ender 3 v2

bsinger99, 2022-05-18:
Hi! I recently purchased a Hermit Crab & an H2 extruder for my Ender 3 v2. I'm having a problem determining how to connect the Hermit Crab to my 4.2.7 motherboard. The provided wiring harness has several unlabeled connectors and at least two (FLD & EO) which I haven't been able to determine what those acronyms refer to.

Also, the concern is that the manual has a picture of the PCB board interfaces connector wiring which doesn't match mine. The PCB-F illustration indicates that the BL TOUCH connector connects with the orange wire on the outside - mine connects with the orange wire at the other end. (Possibly it's just a mistake in the picture?) 

Are there any instructions or a resource you can refer me to help me out?

aidbish, 2022-05-21:
FLD = filament detection (u can use a BTT smart filament detector
E0 = Extruder motor
the black and red floating wires are the extruder heater

rhawley88, 2023-04-03:
> FLD = filament detection (u can use a BTT smart filament detector E0 = Extruder motor the black and red floating wires are the extruder heater

I know this is a year old but how do you use the smart sensor looking for a replacement cable and haven't found. Or do I have to learn to crimp those micro connectors.  Which is fine I just have any of them in stock.