bigtreetech/HermitCrab #27: Hermit Crab 2 and Failed Probe port

5teveO, 2024-04-15:
This may be a bit long winded.. but here goes.

I got a hermit crab V2 canbus from BTT a few weeks back. Flashed it with Katapult and then klipper and all was good from a firmware point of view. I have spent a bit of time building the rest of the machine up and there was a lot of wiring to do as there are a lot of btt external 5160 drivers, PSU control etc.  Intention is to have a biqu microprobe running on the tool heads too (im coming from a canbus hermit crab 1)

Finally got to power the machine on and start configuration.. that all went fine too. Config was easy as it was a swap from an skr1.4t and hermit crab v1 canbus to an octopus pro 1.1 and hermit crab V2 canbus.. 

Started testing the microprobe - which worked very well on previous Hermit crab V1 - and I could not get it to deploy or to retract - which was strange as it worked fine with an identical config (excpet with pins changed to suit HC v2) - swapped probes, cables and even tool plates to see if any of those was an issue. Nothing.

Confirmed 5v was present, and then started testing the GPIO pins so did the query_probe in klipper with the following results - I did this for both GPIO24 being set up as the sensor pin and also with GPIO23 being setup as sensor pin
GPIO24 open -  triggered
GPIO24 shorted to ground -  triggered
GPIO23 open -  triggered
GPIO23 shorted to ground -  triggered

No change in state for either GPIO pins.

To confirm I also set up a BLTouch - same thing - on power up - self tests - then pin up / down does nothing and query probe shows open regardless of whether the pin is up or down.

Would it be possible to give me some guidance as to what else could be wrong?

With no schematics available (nor step files etc) I cannot really do any more troubleshooting.

I've been guided though this by JamesH on the Klipper discord.. @bigtreetech 

If anymore info is required.. please advise asap as I need to get this machine up and running.

Steve

5teveO, 2024-04-15:
Ok i have investigated this a bit more by testing the GPIO's as best I can. It appears that GPIO22,23 and 24 are all non operational - I cannot heat the hot end and deploy / sense the probe.

GPIO 27/28 work and show correct temps
GPIO14 runs FAN1 ok
CAN communicates ok as far as I know so thats GPIO4/5
LIS2DW communicates too - so thats GPIO0/1 working.
I cant prove motor at the moment but will try to do that at some point. 

Where do I go from here. Pretty Sad about it as the HC1 Canbus has been brilliant.

Steve

LancAir390, 2024-04-22:
Were you able to solve your issue?

Like you I had my microprobe working great with Hermit Crab CANbus v1, but have had no luck with Hermit Crab Can 2.

As an aside my hot end does heat up correctly. 

I may have to return this one to Amazon marked as defective...

5teveO, 2024-04-23:
> Were you able to solve your issue?
> 
> Like you I had my microprobe working great with Hermit Crab CANbus v1, but have had no luck with Hermit Crab Can 2.
> 
> As an aside my hot end does heat up correctly.
> 
> I may have to return this one to Amazon marked as defective...

Nope - Conclusion was that the 3 gpio pins are dead for some reason. I received an emal from @bigtreetech Last Wednesday - to which I replied to.. and so far nothing - no response. As i bought it from the Biqu site I dont have the option to just return it, so If I dont get a response in a a few more days - or at least an update from them - I'll be exploring doing a chargeback from the credit card. I cant have my printer down for much longer and dont expect a week between emails.

I'd see if you can do some of the testing I did to prove the GPIO is dead. Did you copy the probe stuff from your HC1 config and change the pin numbers? Is your Microprobe a v1 or v2? Have you tested the probe on another machine? Happy to take a look if it helps... but i assume you know what you are doing... !

lotfinoluffy, 2024-05-26:
have the same issues, so think the board are faulty

lotfinoluffy, 2024-05-26:
have the same issues, so think the board are faulty, on both boards that ive ordered, and as usual no reply from the great renowned btt (add sarcasm here)
just wondering @5teveO if you happen to test the filament runout sensor since i didnt see the pin number in github or the config file, thanks

kasuke5, 2024-09-09:
Hello there, I find the documentation really not clear at all.
I've struggled to make everything works and like all of you I came from Hermitcrab1 which was working ... Until I don't really know what happened, but I tried to plug in board to the main one and the hooks just touch the circuit board and I think burn the main card... (and you can't buy this card anywhere, smart move btt)

Anyway, I've struggled to understand that you actually have a switch button for thermal sensor, even what's inside the package seems to be less than the v1, you only have cables, no jumper, no crimping tools etc ...

So my 3Dtouch was working fine with v1, now it only can pin_up and down but cannot read the sensor values, resulting with no way to home.

I tried using the provided cables, but it's the exact same than previous one.

looxonline, 2024-09-10:
> To confirm I also set up a BLTouch - same thing - on power up - self tests - then pin up / down does nothing and query probe shows open regardless of whether the pin is up or down.

May I suggest that you try to use an endstop port for the probe trigger value instead of the probe pin. I believe that may work.

looxonline, 2024-09-10:
> This may be a bit long winded.. but here goes.
> 
> I got a hermit crab V2 canbus from BTT a few weeks back. Flashed it with Katapult and then klipper and all was good from a firmware point of view. I have spent a bit of time building the rest of the machine up and there was a lot of wiring to do as there are a lot of btt external 5160 drivers, PSU control etc. Intention is to have a biqu microprobe running on the tool heads too (im coming from a canbus hermit crab 1)
> 
> Finally got to power the machine on and start configuration.. that all went fine too. Config was easy as it was a swap from an skr1.4t and hermit crab v1 canbus to an octopus pro 1.1 and hermit crab V2 canbus..
> 
> Started testing the microprobe - which worked very well on previous Hermit crab V1 - and I could not get it to deploy or to retract - which was strange as it worked fine with an identical config (excpet with pins changed to suit HC v2) - swapped probes, cables and even tool plates to see if any of those was an issue. Nothing.
> 
> Confirmed 5v was present, and then started testing the GPIO pins so did the query_probe in klipper with the following results - I did this for both GPIO24 being set up as the sensor pin and also with GPIO23 being setup as sensor pin GPIO24 open - triggered GPIO24 shorted to ground - triggered GPIO23 open - triggered GPIO23 shorted to ground - triggered
> 
> No change in state for either GPIO pins.
> 
> To confirm I also set up a BLTouch - same thing - on power up - self tests - then pin up / down does nothing and query probe shows open regardless of whether the pin is up or down.
> 
> Would it be possible to give me some guidance as to what else could be wrong?
> 
> With no schematics available (nor step files etc) I cannot really do any more troubleshooting. Is it possible the GPIO designation is wrong?
> 
> I've been guided though this by JamesH on the Klipper discord.. @bigtreetech @Alan.Ma
> 
> If anymore info is required.. please advise asap as I need to get this machine up and running.
> 
> Steve

Did you have a pull up enabled on GPIO23 when running this test?

vogll, 2025-02-18:
I can confirm that the Hermit Crab V2 Canbus is distributed with a defective BLTouch circuit. I have sent four emails to support, including a video and a description of the problem. BIQU support confirmed the defective circuit and promised to send a new one. But after three months... nothing. BLTouch is not supported in this version of the Hermit Crab. Don't buy it!

bigtreetech, 2025-02-18:
> I can confirm that the Hermit Crab V2 Canbus is distributed with a defective BLTouch circuit. I have sent four emails to support, including a video and a description of the problem. BIQU support [...]

BIQU here. Can confirm that the circuit is not defective. Please post pics of your removable board(s) and in specific the area where the probe port is located.

vogll, 2025-02-18:
Behaviour of the probe is the same as described at the previous postsby SteveO. 

Sensor_pin of the Hermit Crab doesn't read properly BLTouch output (Sensor pin). After sending homing Command (G28) BLtouch retract sensor but after some seconds Klipper return errors

8:37 BLTouch failed to raise probe
8:37 Failed to verify BLTouch probe is raised; retrying.
8:37 Failed to verify BLTouch probe is raised; retrying.

Here is the another communication from the Clipper

8:40 probe: TRIGGERED
8:40 QUERY_PROBE
8:39 Sending BLTOUCH_DEBUG COMMAND=touch_mode
8:39 BLTOUCH_DEBUG COMMAND=touch_mode
8:39 Sending BLTOUCH_DEBUG COMMAND=pin_down
8:39 BLTOUCH_DEBUG COMMAND=pin_down
8:39 probe: TRIGGERED
8:38 QUERY_PROBE

Here is sniped of my Klipper config file

[bltouch]
sensor_pin:  ^hermitcrab2:gpio24
control_pin: hermitcrab2:gpio23
z_offset : 2.41

[image]
[image]
[image]

Thijskunst, 2025-03-09:
Im having the same issue on my hermit crab v2. It seems that the flyback diode  D3 on all 3 of my removable tool boards have failed open circuit. D3 on my unused spare board has been open circuit from the factory.  D3 is there to protect the gpio pin from the reverse voltage spike from turning off the  probe.  

I don't think it's a problem with the circuit, I think its an issue with cheap, dodgy parts that have a high failure rate.

I've contacted BTT on AliExpress, where I bought my hermit crab v2, but haven't heard back from them yet.

vogll, 2025-04-13:
After 4 months Biqu sent me new Hermit Crab board. BLtouch working as expected. However, the correct settings are not the ones posted by BigTreeTech, but rather:

```
[bltouch]
sensor_pin: ^hermitcrab2:gpio24
control_pin: hermitcrab2:gpio23
```

lotfinoluffy, 2025-07-24:
@bigtreetech doesn't change anything, and the ... if it was the correct one why is it the same as @vogll 
any of those configs and you get the probe on the triggered state whether the pin is up or down 

ordered 2 units and what a PAAAAAAAIN it was and none of them is working and the support dare with no shame shift every single blame on me when its your selling faulty boards

also blotch or eddy coil, the probe also gives a Unable to obtain 'i2c_read_response' response or a shutdown: I2C START READ NACK with your config and no matter what it doesn't work